rivm-syso/Analyse-Together #10: Differences between sensor and reference station

elma13, 2022-05-24:
Do we want to make a difference between sensor and reference station?
In the current tool the differences are:

- The shape of the marker
- The color and line type in the plots. Each sensor get its own colour, while each reference station has the same colour but different line type
- Not every plot was available for the reference stations, only the plots where comparison with the sensors was directly
- Grouping was only possible for the sensors

Do we still think this usefull? Then meta data is needed to know which station is a sensor and which is a reference station.

elma13, 2022-05-31:
We like to have the difference in the plotting, color and linetype.
In the part of loading and getting the data, the meta data needs to be read. Then the type (ref station/sensor) can be added to the location table. Also the label for the station name can be stored there.

jspijker, 2022-07-27:
This issue depends on #27

closed 2022-09-15 by Hendricxw