rivm-syso/Analyse-Together #39: Fix db connection error

jspijker, 2022-07-14:
Error on local machine, not always reproducible on other machines. Why?

Hendricxw, 2022-07-15:
For me, with the previous version from https://github.com/rivm-syso/ATdatabase/blob/develop/R/download_data.R I now get another error: 

DEBUG [2022-07-15 11:18:40] downloading measurements for some 269, 1/45
DEBUG [2022-07-15 11:18:40] downloading data for station 269 for time range 20220101 -  20220204
[1] "wget -O - --no-check-certificate --post-data='stns=269&start=2022010101&end=2022020424&vars=DD:FF:T:U' https://www.daggegevens.knmi.nl/klimatologie/uurgegevens"
--2022-07-15 11:18:40--  https://www.daggegevens.knmi.nl/klimatologie/uurgegevens
Resolving www.daggegevens.knmi.nl (www.daggegevens.knmi.nl)... 63.32.154.219, 52.214.113.7, 54.194.1.42
Connecting to www.daggegevens.knmi.nl (www.daggegevens.knmi.nl)|63.32.154.219|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/plain]
Saving to: ‘STDOUT’

     0K ..                                                     77.8M=0s

2022-07-15 11:18:40 (77.8 MB/s) - written to stdout [2119]

[1] "KNMI: ruwe data opgehaald"
[1] "KNMI: data omgezet naar dataframe"
[1] "KNMI: metadata omgezet naar dataframe"
[1] "KNMI: metadata en data gecombineerd"

**Error in `$<-.data.frame`(`*tmp*`, "station", value = "KNMI_") :
replacement has 1 row, data has 0**

jspijker, 2022-07-18:
I can;t reproduce the connection error, database connection appears to work fine. I have no clue what happened but probably it was some hickup with library versions.

@Hendricxw above error looks like something which is not database related.

Hendricxw, 2022-07-18:
Unfortunately I still get the error message.

SanneYpenburg, 2022-07-19:
Unfortunately, I have a new error as well (not at the KNMI part, that still works fine, but at the downloading the sensor data part):

TRACE [2022-07-19 10:11:06] skipping store stream data info for AMF_pm008
DEBUG [2022-07-19 10:11:06] downloading measurements for station AMF_pm015, 1/18

 Error in h(simpleError(msg, call)) : 
  error in evaluating the argument 'conn' in selecting a method for function 'dbQuoteLiteral': argument "conn" is missing, with no default

jspijker, 2022-07-26:
Resolved with https://github.com/rivm-syso/Analyse-Together/commit/9421501350f2b7f0112b9a38346c75c3a8b22b51

closed 2022-07-26 by jspijker