rivm-syso/Analyse-Together #61: Choices in select data menu must start data download

jspijker, 2022-09-07:
Based on the choices in the select data tab and pull down menu's. a data download should start en data must be added to the database.  Therefore we have to:

 * [ ] figure out how selected values are communicated to involved modules (or new module)
 * [ ] create interface with function which initiates download
 * [ ] connect it al together

And while we're on it anyway, also clean up code in the database_fun file. 
See also #25

jspijker, 2022-09-22:
Now rewrite function in https://github.com/rivm-syso/Analyse-Together/commit/8c074af636bd5eb6164f6377f6a8b443070432d1 into background function that downloads sensor + meteo + AQ ref. station. Store locations in calling function.

jspijker, 2022-10-05:
Closed with PR #75

closed 2022-10-05 by jspijker

elma13, 2022-10-18:
Add all functionalities inside the download button:

- download the meta data for the project/municipality
- get list of the station names
- create the queue
- download all sensor data