rivm-syso/Analyse-Together #67: Text and labels in the front end to Dutch -> make use of javascript package thing

elma13, 2022-09-26:
The first approach is in Dutch, however use the library to store the text. Then we can translate it later easily.

Internationalisatie: https://appsilon.github.io/shiny.i18n/ en https://github.com/Appsilon/shiny.i18n

elma13, 2022-09-26:
Keys in the code in English, in the JSON file the translation to Dutch

elma13, 2022-09-26:
The keys are written with a tag, in the json the whole text in dutch and english.

guidelines for the keys: 

- Information and longer text starts with "expl_"
- All elements for selection starts with "sel_"
- All titles and headers starts with "title_"
- All elements of the plots starts with "plot_"

elma13, 2022-09-26:
It doesnot work in every function, for example the add_metadata module if you try to change the names of the columns:

 try(datatable(data_merged(),colnames = c(i18n$t("word_obs") = "n_obs", "test2" = "station_type"),

[image]

elma13, 2022-09-26:
TODO: a nice place and flags for the different languages layout

jspijker, 2022-10-05:
closed with PR #71

closed 2022-10-05 by jspijker