rivm-syso/Analyse-Together #79: Error while downloading data

jspijker, 2022-10-07:
Both downloading a project or municipality results in an error:
```
DEBUG [2022-10-07 10:30:11] GetSamenMetenAPIinfo: Data received from https://api-samen
meten.rivm.nl/v1.0/Things?$filter=%28properties%2Fproject+eq%27Luchtclub%27%29&$expand
=Locations%2CDatastreams&$top=200&$skip=200
Error in names(datastream_list) <- kit_id_overview : 
  'names' attribute [400] must be the same length as the vector [200]
```
Happens in the develop branch, demo download script

jspijker, 2022-10-07:
Error boils down to an error in the samanapir package:

```
samanapir::GetSamenMetenAPIinfoMuni("599")                                                                                                                 
INFO [2022-10-07 14:51:15] GetSamenMetenAPIinfo: Get data from url: https://api-samenmeten.rivm.nl/v1.0/Things?$filter=(properties/codegemeente%20eq'599')&$expand=Locations,D
atastreams                                                                                                                                                                    
TRACE [2022-10-07 14:51:15] GetAPIDataframe: Getting url https://api-samenmeten.rivm.nl/v1.0/Things?$filter=(properties/codegemeente%20eq'599')&$expand=Locations,Datastreams 
TRACE [2022-10-07 14:51:16] GetAPIDataframe: Getting url, got result 200                                                                                                      
DEBUG [2022-10-07 14:51:16] GetSamenMetenAPIinfo: Data received from https://api-samenmeten.rivm.nl/v1.0/Things?$filter=(properties/codegemeente%20eq'599')&$expand=Locations,
Datastreams                                                                                                                                                                   
TRACE [2022-10-07 14:51:18] GetSamenMetenAPIinfo: getting next page ...                                                                                                       
INFO [2022-10-07 14:51:18] GetSamenMetenAPIinfo: Get data from url: https://api-samenmeten.rivm.nl/v1.0/Things?$filter=%28properties%2Fcodegemeente+eq%27599%27%29&$expand=Loc
ations%2CDatastreams&$top=200&$skip=200                                                                                                                                       
TRACE [2022-10-07 14:51:18] GetAPIDataframe: Getting url https://api-samenmeten.rivm.nl/v1.0/Things?$filter=%28properties%2Fcodegemeente+eq%27599%27%29&$expand=Locations%2CDa
tastreams&$top=200&$skip=200                                                                                                                                                  
TRACE [2022-10-07 14:51:18] GetAPIDataframe: Getting url, got result 200                                                                                                      
DEBUG [2022-10-07 14:51:18] GetSamenMetenAPIinfo: Data received from https://api-samenmeten.rivm.nl/v1.0/Things?$filter=%28properties%2Fcodegemeente+eq%27599%27%29&$expand=Lo
cations%2CDatastreams&$top=200&$skip=200                                                                                                                                      
Error in names(datastream_list) <- kit_id_overview :                                                                                                                          
  'names' attribute [400] must be the same length as the vector [200] 
```
@elma13 any clue? Looks likes there is something wrong with adding the next page.

elma13, 2022-10-17:
@jspijker, thanks for finding this bug! I solved the problem, can you check if it works again? (Still in branch issue2)

Strange other thing is that the logger also caused for problems, so I left some logging out...

jspijker, 2022-10-26:
Nope, doesn't work. I update the samanapir pkg (Issue_2 branch) and still got the same error. @elma13, I used the same code as above.