rivm-syso/Analyse-Together #106: Add page with que download info

elma13, 2022-10-31:
To get more insights in the downloading process, create a page where you can view the que. This is for developing interesting, no need for regular users, yet.

elma13, 2022-10-31:
@jspijker , in the queue the status " duplicated" is used when a sensor is already in the list. What does that mean? Will the job be executed as normal?

jspijker, 2022-11-10:
The duplicate status means that the job is 'on hold' It will not be queued because another job, for the same sensor,  is already running or on the waiting list. The scheduler checks the duplicate state every time, when there is no job running or waiting, the duplicate state is changed into waiting.

closed 2023-03-09 by elma13