rivm-syso/Analyse-Together #147: When deselecting a station, only the colour changes, the barplot stays. Something with the filt measurements?

elma13, 2023-01-12:


elma13, 2023-01-12:
Is this still the case?

elma13, 2023-03-09:
This is for all plots, when deselecting all, then the graph stays showing the last one.

elma13, 2023-03-10:
The graphs should be redrawn after deselecting all sensors. And maybe also after entering the Select-tab. So if you enter the visualisation tab, no sensors are selected and no plot is drawn.

elma13, 2023-03-10:
The selection on the map is already removed, however the selection boxes are still shown. Can those also be cleared?

elma13, 2023-05-11:
If you change your selection in the select-tab, then the previous sensors are still on the map and in the figures. If you select a sensor, then the tool crashes

jspijker, 2024-02-26:
closed, superseded by #281

closed 2024-02-26 by jspijker