rivm-syso/Analyse-Together #368: Record the status of LML and KNMI API requests

jspijker, 2024-08-12:
The LML and KNMI API's are often not available for some reasons. A data request to these API's returns in such cases an error or time-out. However in the ATdb the time range is recorded with no data. This result in that no data wil be added if the API is online again.

Think of something that records the result of the API's and in the case of an error, do not add the time range to the caching table. Or, postpone the data request and reloads it when the schedule is running.

jspijker, 2024-08-12:
I can't reproduce this situation. If I force errors in the API, the samanapir functions fail and produce an regular error. The download sub-process halts and no data is stored. So, the next time a sensor is requested (and the API is up again), the missing LML and KNMI data is downloaded. 

While the errors are ugly, the test results indicate that app behaves as intended. To catch the errors and create a postponed data request is quite some effort and I think the use case is limited. My suggestion is to close this issue and leave things as they are. @elma13, @Hendricxw : what do you think?

Hendricxw, 2024-08-13:
It is hard to estimate how frequent this might occur. In worse case: data might not be correctly downloaded, until the time-span is changed, right?!

jspijker, 2024-08-26:
If the data is requested again, independent of time span, it will be downloaded (if the API is available again)